pvlib/pvlib-python #2407: v0.11.3 (or 0.12.0?) planning

kandersolar, 2025-03-14:
The [v0.11.3 milestone](https://github.com/pvlib/pvlib-python/milestone/49) is cleared out and the release is ready to finalize.  Along the way, we've merged some PRs including what we're calling breaking changes:

https://github.com/pvlib/pvlib-python/blob/40a213b7417400e6c974c17115636d1a0312dbfd/docs/sphinx/source/whatsnew/v0.11.3.rst?plain=1#L7-L14

Our versioning strategy says that including such changes mean we should call this release 0.12.0 instead of 0.11.3.  However, we have several deprecations marked to be removed in 0.12: https://github.com/pvlib/pvlib-python/issues/2396

@pvlib/pvlib-maintainer @pvlib/pvlib-triage your input is requested:  should we call this release 0.12.0 and get some final PRs in to address those deprecations?

AdamRJensen, 2025-03-14:
Not really addressing what you're asking but worth mentioning. I plan to implement several breaking changes in the iotools module to bring more consistency. This I hope to have released before the PVSC pvlib tutorial in early June

cwhanse, 2025-03-14:
No objections from me to make the next release v0.12.  I marked #2343 for v0.13 thinking we should deprecate `pytz` in v0.12.

kandersolar, 2025-03-14:
Ok, I moved the 0.11.3 items to 0.12.0, renamed 0.11.4 to 0.12.1, and deleted the 0.11.3 milestone.

Looks like @echedey-ls, @RDaxini, and @AdamRJensen already have PRs started for some of these items.  Let's try to get those and the rest of the 0.12.0 removals done so that 0.12.0 can be released sometime next week.

cwhanse, 2025-03-14:
Do we want to deprecate `pytz` in 0.12?

RDaxini, 2025-03-14:
> * [Remove deprecated `pvsystem.sapm_spectral_loss` #2244](https://github.com/pvlib/pvlib-python/pull/2244)

#2244 is now ready

What do we think about #2368? It was just a suggestion. It's not a big change but, if it would be helpful to use subfolders, I can open a PR in time for this next release.

kandersolar, 2025-03-14:
> What do we think about https://github.com/pvlib/pvlib-python/issues/2368?

+1 from me, but it's not a breaking code change and can easily wait for the next release if time is a constraint

PR #2413 (2025-03-19): Finalize v0.12.0
 - [x] Closes #2407
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`nu [... cut]
closed 2025-03-19 by kandersolar