pvlib/pvlib-python #2412: Remove ``outputformat`` parameter in get_pvgis_hourly and get_pvgis_tmy

AdamRJensen, 2025-03-14:
Supporting the four different ``outputformat`` options in ``get_pvgis_hourly`` and ``get_pvgis_tmy`` is proving to be a rather high maintenance burden without much benefit. I therefore suggest that we altogether get rid of the parameter and simply choose the json option as this is the easiest to parse / require less maintenance and features more or similar metadata as other options.

This issue is already mentioned in #2344 and #2399.

Here's the metadata available when ``outputformat='csv'``
**Header:** 
[image]

**Footer:**
[image]

Here's the metadata available when ``outputformat='json```

[image]

[image]

kandersolar, 2025-03-15:
I think `outputformat='epw'` is worth considering.  It returns two additional data columns: `'aerosol_optical_depth'` and `'temp_dew'`.  It's also even easier to parse than the JSON, as it needs only a call to the `parse_epw` function.

Another question: if we get rid of `outputformat` in `get_pvgis_tmy`, what will happen with `read_pvgis_tmy`?

AdamRJensen, 2025-03-22:
As long as we get rid of ``outputformat='basic'`` in #2416 I'm ok keeping the others for now.

closed 2025-03-22 by AdamRJensen