pvlib/pvlib-python #2414: Prilliman exponential factor sign mistake

mio-pv, 2025-03-21:
**Describe the bug**
Prilliman temperature function seems to have a sign mistake.
in `temperature.py:prilliman(temp_cell, wind_speed, unit_mass=11.1, coefficients=None):`
```
    timedeltas = np.arange(samples_per_window, 0, -1) * sample_interval * 60
    weights = np.exp(-p[:, np.newaxis] * timedeltas)
```
shouldn't it be
```
    timedeltas = np.arange(samples_per_window, 0, 1) * sample_interval * 60
    weights = np.exp(-p[:, np.newaxis] * timedeltas)
```
so that time deltas are positive and exponential factors are negative?

**Additional context**
See publication
[1] M. Prilliman, J. S. Stein, D. Riley and G. Tamizhmani,
       "Transient Weighted Moving-Average Model of Photovoltaic Module
       Back-Surface Temperature," IEEE Journal of Photovoltaics, 2020.
       doi:`10.1109/JPHOTOV.2020.2992351`

adriesse, 2025-03-21:
The time deltas are actually positive.  The -1 is the increment for counting down to zero.

mio-pv, 2025-03-21:
you are right this is a dumb mistake... thanks a lot

closed 2025-03-21 by mio-pv