pvlib/pvlib-python #2420: ERA5 single-point time series database

AdamRJensen, 2025-03-23:
ERA5 has recently launched a [time series-specific database ](https://cds.climate.copernicus.eu/datasets/reanalysis-era5-single-levels-timeseries?) for ERA5 which brings down request times from +10 minutes to a few seconds.

Note, that the service is currently listed as experimental. Here's the info from the ECMWF's website.

> Please be aware that this is an experimental catalogue entry which uses an alternative source for the ERA5 data designed for fast access long-timeseries for a single point, and it is not recommended for use operational systems.
> The entry may be temporarily disabled or completely deprecated at any point and it does not come with the same level of operational support as the [parent entry](https://cds.climate.copernicus.eu/datasets/reanalysis-era5-single-levels). Additionally, the formats and structure of the data files are subject to change.

LinkedIn announcement:

I plan on making a PR in the following week, but we may delay merging until the service becomes permanent.

AdamRJensen, 2025-03-23:
Implementing a wrapper around this API addresses #1264 and in part #1484.

@williamhobbs ICYMI

adriesse, 2025-03-24:
Slight downside of this dataset is that is does not include solar radiation.  😭

AdamRJensen, 2025-03-24:
> Slight downside of this dataset is that it does not include solar radiation. 😭

Thanks for noticing this! I'll contact the development team and see if we can't include this. Feel free to also leave a comment [here](https://forum.ecmwf.int/t/new-dataset-published-in-cds-era5-hourly-time-series-data-on-single-levels-from-1940-to-present/11919).