pvlib/pvlib-python #2422: Tools for Generating Synthetic Irradiance Timeseries

jranalli, 2025-03-27:
In my [solarspatialtools ](https://github.com/jranalli/solarspatialtools) library, I made an [implementation](https://solarspatialtools.readthedocs.io/en/stable/demos/synthetic_clouds_demo.html) of [Lave et al's](https://dx.doi.org/10.1109/PVSC.2017.8366378) synthetic irradiance approach based on stacked multi-scale random cloud fields. 

I wanted to offer to contribute it over here if people would be interested, because my philosophy for that project is to rely on pvlib where possible since it's much more mature and widely adopted, and only to house things that are out of pvlib's scope. In this case, it might be relevant to expansion of the `scaling` package in pvlib, but no offense taken this is too far afield. This code is pretty involved (600+ lines), so it would be a big review, but is relatively self-contained.

mikofski, 2025-03-27:
+1 if it can scale monthly to hourly. What inputs does it require? Any sense of expected downsides? Eg if used to scale hourly to 5-minute what difference versus ground should I be concerned about?

cwhanse, 2025-03-28:
I think this method is insufficiently validated for inclusion in pvlib, in particular, because its likely use in pvlib is to extrapolate a point irradiance measurement to a field of irradiance and to accept each field point as realistic. This is beyond the aims of the developers, and the paper admits as much (emphasis added):

"However, when timeseries are sampled at hundreds of locations (corresponding to the hundreds of different transformer locations on the feeder), the aggregate output is much smoother and **looks more realistic**. As described in section Error! Reference source not found., we have ongoing test to evaluate the need for accurate distributed PV inputs. For analysis such as voltage regulator tap changes, **it may not be important that a single customer be accurately portrayed** because the regulator will only see the aggregate output of several PV systems."

jranalli, 2025-03-28:
I certainly agree with that description of the method's validation and can see the potential for misinterpretation of what the spatial field really represents.

mikofski, 2025-03-29:
Hi @jranalli I might have an alternate method for synthesizing high frequency data, I think it’s an implementation of a popular algorithm. I’ll send it to you to see what you think.

jranalli, 2025-03-29:
Sure I'd love to see it. I'm definitely up for seeing a variety of documented methods. I think a comparative validation of some of the downscaling and/or synthetic irradiance methods would be something useful for a future study.

mikofski, 2025-04-01:
HI Joe, the method I have was written by Patrick Mathiesen based on:
* ["Synthetic generation of high temporal resolution solar radiation
data using Markov models," by Ngoko et al, in Solar Energy (2014)](https://doi.org/10.1016/j.solener.2014.02.026) 
* and ["Stochastic generation of synthetic minutely irradiance time
series derived from mean hourly weather observation data," by Bright et al, in Solar Energy (2015)](https://doi.org/10.1016/j.solener.2015.02.032). 

I have to convert it from matlab, and will test it out and post it as soon as I get the chance.

AdamRJensen, 2026-09-29:
Closing this issue due to lack of progress.

closed 2026-09-29 by AdamRJensen