pvlib/pvlib-python #2425: fit_desoto runtime error

kurt-rhee, 2025-04-01:
**Describe the bug**
Hey all, I am having trouble using the fit_desoto function for this particular longi 540 module.  Parameters are from the CEC database and then converted into units of A/K V/K.  I could switch to a different fitting model if required.  

**To Reproduce**
```

TEMP_REF = 25.0
IRRAD_REF = 1000.0

{'vmp': 41.65, 'imp': 12.96518607, 'voc': 49.5, 'isc': 13.84716319, 'alpha_isc': 0.0052619220122, 'beta_voc': -0.13959, 'cells_in_series': 72, 'eg': 1.121, 'degdt': -0.0002677}

params, optimization_results = pvlib.ivtools.sdm.fit_desoto(
    v_mp=module["vmp"],
    i_mp=module["imp"],
    v_oc=module["voc"],
    i_sc=module["isc"],
    alpha_sc=module["alpha_isc"],
    beta_voc=module["beta_voc"],
    cells_in_series=module["cells_in_series"],
    EgRef=module["eg"],
    dEgdT=module["degdt"],
    temp_ref=TEMP_REF,  # type: ignore
    irrad_ref=IRRAD_REF,  # type: ignore
)
```
```
RuntimeError: Parameter estimation failed:
The iteration is not making good progress, as measured by the
 improvement from the last five Jacobian evaluations.
```
**Expected behavior**
Parameters calculated with no runtime error

**Versions:**
 - ``pvlib.__version__``: 0.11.2
 - ``pandas.__version__``: 2.2.3
 - python: 3.11.10

cwhanse, 2025-04-01:
The error is from `scipy.optimize`, and indicates that the objective function is "stuck" for some reason.

The `fit_desoto` function implements an algorithm in Duffie and Beckman. I've observed it to struggle with modules with high current, likely because of the way the initial guess is made. In this example, initial Rs is 0.016 which is too low, and initial Io is 2.5e-7 which is likely too large. The initial guess follows the published reference.

I would try `fit_cec_sam` instead if you have PySAM installed.

kandersolar, 2025-04-01:
In this particular case, convergence can be had by supplying a custom starting point: `init_guess=dict(Rs_0=0.2, Io_0=1e-13, a_0=1.5)`.  This requires pvlib>=0.11.2.

mikofski, 2025-04-01:
For Kurt and others, here’s a [discussion on handling convergence issues when fitting PV module parameters](https://github.com/SunPower/PVMismatch/discussions/143).

kurt-rhee, 2025-04-02:
Wow thank you so much to all of you, that is all very informative.  

@cwhanse I've downloaded some of the osti papers in the references for further reading
@kandersolar how did you know that those initial guess parameters would work well?
@mikofski that thread is really helpful for understanding what is happening under the hood and what can be done about it

closed 2025-04-02 by kurt-rhee

kandersolar, 2025-04-02:
> @kandersolar how did you know that those initial guess parameters would work well?

Nothing clever, I'm afraid.  I edited the `fit_desoto` function to print the starting values, determined the ones I thought seemed like bad values (as @cwhanse did), and then tried a few combinations of more reasonable values until it converged successfully.

markcampanelli, 2025-04-02:
@kurt-rhee Some more ideas at https://github.com/markcampanelli/pvfit/blob/main/pvfit/modeling/dc/single_diode/equation/simple/inference_ic.py

kurt-rhee, 2025-04-03:
Awesome this is really nice too thank you!

joseph-sch, 2025-12-31:
I believe it’s worth mentioning that PR #2563 (closing #2562) introduces a non-iterative method, `pvlib.ivtools.sdm.fit_desoto_batzelis` (from Batzelis, 2017), for estimating single-diode model parameters from datasheet values:

https://pvlib-python.readthedocs.io/en/latest/reference/generated/pvlib.ivtools.sdm.fit_desoto_batzelis.html#pvlib.ivtools.sdm.fit_desoto_batzelis

The changes are announced in the upcoming version `0.13.2`: https://pvlib-python.readthedocs.io/en/latest/whatsnew.html#enhancements