pvlib/pvlib-python #2438: Implement a new temperature model and fitting methods

adriesse, 2025-04-15:
I published a comprehensive temperature model that pulls together handling of dynamics and radiative losses at EU-PVSEC 2024. The paper title doesn't mention a new model, but I got a bit carried away while working on the parameter extraction between the time of the abstract submission and conference.  🙂

https://doi.org/10.4229/EUPVSEC2024/4BO.6.6

kandersolar, 2025-04-15:
@adriesse could you email me a copy?  I don't have access to the proceedings.  Thanks!

adriesse, 2025-04-15:
> [@adriesse](https://github.com/adriesse) could you email me a copy? I don't have access to the proceedings. Thanks!

I think you everyone has access, but you have to create an account.  This also works:

https://www.researchgate.net/publication/388269643_PV_Module_Operating_Temperature_Reliable_Extraction_of_Model_Parameters_from_Dynamic_Field_Data