pvlib/pvlib-python #2446: `iotools` reference page needs some organization

kandersolar, 2025-04-23:
The page in question: https://pvlib-python.readthedocs.io/en/latest/reference/iotools.html

The main listing has 40-50 functions.  Too many!  Let's split it up into smaller listings.

The question is, how should it be split?  I suggest organizing by data provider, i.e. separate listings for `PVGIS`, `NSRDB`, `Solcast`, etc.

cwhanse, 2025-04-23:
> organizing by data provider

gets my vote

wholmgren, 2025-04-23:
At one point we (or at least I) had the thought that the many iotools functions would no longer be imported into the base iotools namespace and users would need to reference the functions in the specific module. I think modifying the documentation to reflect that would be a good first step. I think given the structure of the subpackage, that's a long way of saying +1 for organizing by data provider.

adriesse, 2025-04-24:
There was also some talk of separating iotools from pvlib-python.  If that happens the iotools documentation would naturally also be restructured.