pvlib/pvlib-python #2448: Glossary update: surface/solar zenith/azimuth angles, and others

RDaxini, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
Some longer definitions and conventions are being removed from parameter descriptions in #2311

**Describe the solution you'd like**
Migrate definitions and conventions over to the [nomenclature page](https://pvlib-python.readthedocs.io/en/stable/user_guide/nomenclature.html)

See thread below for examples.

**Additional context**
Let's agree on some definitions here, then I'll add them to the glossary.
Note: some conventions (e.g. angle ranges or irradiance limits) might be function specific.
Related discussion/comments: Issue #1421, [comment in #1421](https://github.com/pvlib/pvlib-python/issues/1421#issuecomment-1076860942), [Comment in #2311](https://github.com/pvlib/pvlib-python/pull/2311#discussion_r2056805803), #2284

RDaxini, 2025-04-25:
Original definitions from `irradiance.py` with some suggested modifications from me.

```
surface_tilt
Surface tilt angle, must be >=0°  and <=180°. The surface tilt angle describes
the inclination of the PV panel and is defined as degrees from the horizontal
such that a surface facing up would have a surface tilt of 0°, and one facing
the horizon would be 90°.  [°]

surface_azimuth
Surface azimuth angle, must be >=0° and <=360°. This angle describes the
horizontal orientation of the PV panel relative to a reference direction, and
the convention is defined as degrees east (clockwise) of north. For example,
North = 0°, South=180° East = 90°, West = 270°.  [°]

solar_zenith
Apparent (refraction-corrected) zenith angle, must be ≥0° and ≤180°. The solar
zenith angle describes the position of the sun relative to the vertical and is defined
as the angle between the sun’s rays and a vector pointing straight up from the
observer. [°]

solar_azimuth
Solar azimuth angle, must be ≥0° and ≤360°. The solar azimuth angle describes
the sun’s position along the horizon and is defined as degrees east (clockwise)
of north. For example, North = 0°, East = 90°, South = 180°, West = 270°. [°]

Some of these would be new definitions, some of these exist in the glossary but need to be updated with more detail.
There are others. I will update this list...

kandersolar, 2025-05-08:
A couple suggestions:

- `surface_azimuth`: remove "for example", as that could be interpreted as indicating that the user could choose another convention if they like.  Also make clear that east of north is _pvlib's_ convention, and other tools use different conventions.
- `solar_zenith`: add something like ```This is the complement of :term:`solar_elevation`.```

williamhobbs, 2025-05-08:
Is this open issue relevant, https://github.com/pvlib/pvlib-python/issues/1976?