pvlib/pvlib-python #2452: wrong URL for PSM v3.2.2

mikofski, 2025-05-04:
https://github.com/pvlib/pvlib-python/blob/216386ef2f2fcdcaf1b3ecb5913abdf54e9eabaf/pvlib/iotools/psm3.py#L15

According to [NREL Developer](https://developer.nrel.gov/docs/solar/nsrdb/psm3-2-2-tmy-download/) & [pvlib](https://pvlib-python.readthedocs.io/en/v0.10.5/reference/generated/pvlib.iotools.get_psm3.html#pvlib.iotools.get_psm3) docs, this should be: `https://developer.nrel.gov//api/nsrdb/v2/solar/psm3-2-2-tmy-download.format?parameters`.

mikofski, 2025-05-04:
Doesn't really matter, b/c I can't get either NSRDB endpoint to work consistently for past 2 days anyway. See https://github.com/NREL/developer.nrel.gov/issues/382#issuecomment-2849482949 .

kandersolar, 2025-11-18:
Closing since the PSM3 API no longer exists and we have removed the relevant functions.

closed 2025-11-18 by kandersolar