pvlib/pvlib-python #2453: Add REST2 clear-sky model

AdamRJensen, 2025-05-07:
The REST2 clear-sky model is developed by Chris Gueymard, who has expressed serious interest in contributing this to pvlib. Notably, the REST2 model is used to generate the NSRDB, so that alone makes it interesting for pvlib in my opinion.

The most obvious place to add it is in the clearsky.py module, although we may consider organizing clear-sky functions into a folder with .py files for each clear-sky function.

Note, NREL has a [Python implementation](https://github.com/NREL/rest2), however, that is outdated. Similarly, Jamie Bright ported version v5 to python [here](https://github.com/JamieMBright/clear-sky-models/blob/master/python/clearSkyRadiation_REST2.py).
 
It would be great if you could leave a thumbs up if you think this would be a useful addition, and otherwise, any comments are appreciated concerning implementation, etc.

The best reference for REST2 is [10.1016/j.solener.2007.04.008](https://doi.org/10.1016/j.solener.2007.04.008) (refers to v5). Later validation papers related to both v5 and v9.1 are: [10.1016/j.rser.2019.04.006](https://doi.org/10.1016/j.rser.2019.04.006)
and [10.1016/j.rser.2020.110087](https://doi.org/10.1016/j.rser.2020.110087). There is no paper related to the latest v11.5 yet, but Chris is working on it.

ICYMI @adriesse

cwhanse, 2025-05-07:
While REST2 may not provide broad value to PV modelers, there are enough users of pvlib outside of PV modelers that including REST2 has value.

I'm neutral about splitting up clearsky.py.  We'd need to keep track of REST2 versions and stay faithful to a version's reference, I think.

mikofski, 2025-05-08:
Related: https://github.com/pvlib/pvlib-python/issues/71#issuecomment-187992215

mikofski, 2025-05-08:
Unrelated I would love to see an all sky model like farms-nit

adriesse, 2025-05-10:
> While REST2 may not provide broad value to PV modelers, there are enough users of pvlib outside of PV modelers that including REST2 has value.

I actually think clear sky models are very valuable for PV modelers, although some of them may not know that (yet)!

williamhobbs, 2025-11-18:
I like this idea. 

> I actually think clear sky models are very valuable for PV modelers, although some of them may not know that (yet)!

Agreed!