pvlib/pvlib-python #2455: New default solar position algorithm

kurt-rhee, 2025-05-16:
**Is your feature request related to a problem? Please describe.**

[Solar Position Documentation](https://pvlib-python.readthedocs.io/en/stable/reference/solarposition.html
https://pvlib-python.readthedocs.io/en/stable/reference/generated/) recommends using the default solar position algorithm.  Should we update to a faster algorithm given @AdamRJensen's poster at this year's PVPMC which shows that most algorithms have minimal impact on energy regardless of run-time?

With permission I can edit this comment and add a photo of the poster.

**Describe the solution you'd like**
A fast implementation of the SG2, USNO or NOAA solar position algorithms set as the default model for the `get_solarposition` wrapper.

**Describe alternatives you've considered**
We could leave the default as `nrel_numpy`

**Additional context**
- Increased run-time speed of pvlib reduces carbon output of running models (no matter how small or large that might be)
- I am happy to re-implement one of the above models in Rust with python bindings and open source the rust backend.  I know that this comment will open up a can of worms.  I think one way to go about this is to implement the model twice, once in python as a reference for the community and once in rust/julia/c++ etc. for use in practice.  Users would be able to select which model to use at their prerogative.

williamhobbs, 2025-05-17:
I like the idea. A picture of the poster (with permission) would definitely be nice.

mikofski, 2025-05-17:
Solpos is a great algorithm as fast as usno but as accurate as spa and it already has a c implementation. Try `pip install solarutils` or see https://sunpower.github.io/SolarUtils/

kurt-rhee, 2025-05-17:
@mikofski that is awesome, would it be feasible to add the algorithm to pvlib to make it easier for newcomers to use?  I wouldn't mind doing the work.

mikofski, 2025-05-17:
I would appreciate a Python version of SOLPOS. Or pvlib can also import optional dependent, fire example perhaps consider [sg2](https://pypi.org/project/sg2/)

Past concerns were
* license
* cython/binary maintenance

[SPECTRL2 addition by Kevin](https://github.com/pvlib/pvlib-python/pull/1062) resolved these concerns by reimplementing natively.

Related / past:
* https://github.com/pvlib/pvlib-python/pull/730
* https://github.com/pvlib/pvlib-python/pull/2167
* https://github.com/pvlib/pvlib-python/pull/2169
* https://github.com/pvlib/pvlib-python/pull/2152
* https://github.com/pvlib/pvlib-python/issues/1906
* https://github.com/pvlib/pvlib-python/discussions/1308
* https://github.com/pvlib/pvlib-python/discussions/1714
* https://github.com/pvlib/pvlib-python/pull/1443
* https://github.com/pvlib/pvlib-python/issues/1060

Purpose of listing these just for reference

adriesse, 2025-05-18:
Are there issues here?

1. Changing the default algorithm
2. Adding another algorithm

I admit SPA sometimes takes longer than I would like, but I can cache the results if needed.  Consistency is most important to me.

Here's a slide from software comparisons I did (*what!? it's already*) 6 years ago.

[image]

kandersolar, 2025-05-18:
> Should we update to a faster algorithm given @AdamRJensen's poster at this year's PVPMC which shows that most algorithms have minimal impact on energy regardless of run-time?

A paper is in the works with more detail on this topic.  Note that we already have a faster alternative to SPA (`pvlib.solarposition.ephemeris`), but quoting https://github.com/pvlib/pvlib-python/issues/2323#issuecomment-2517919587:

> A bit of context: @AdamRJensen, @IoannisSifnaios, and I have a project underway comparing solar position algorithms. We plan to contribute implementations of one or more algorithms to pvlib-python, depending on the results of the comparison. We expect some of them to be better than `ephemeris` in all respects (speed, accuracy, period of validity, clear reference).

#2152 and #2167 are candidates, but those PRs are on hold for now while we work towards clear recommendations for choice of algorithm. 

> Solpos is a great algorithm as fast as usno but as accurate as spa

I do not think this is true, unless "accurate as spa" means "produces practically equivalent solar energy simulations", which is also the case for half a dozen other algorithms.