pvlib/pvlib-python #2456: "View on github" links not working

cwhanse, 2025-05-17:
Some "View on github" links return "page not found" or direct to lines different from the source function.

**To Reproduce**
I tested three links:

For [pvsystem.v_from_i](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.v_from_i.html) the "View on github" link directs to [these lines](https://github.com/pvlib/pvlib-python/blob/main/pvlib/pvsystem.py#L2603-L2682) in the `max_power_pt` function.

For [pvsystem.max_power_point](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.max_power_point.html) the "View on github" link directs to [these lines](https://github.com/pvlib/pvlib-python/blob/main/pvlib/pvsystem.py#L2542-L2600).

For [ivtools.sdm.pvsyst_temperature_coeff](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.ivtools.sdm.pvsyst_temperature_coeff.html) the "View on github" link returns page not found.

echedey-ls, 2025-05-17:
The reason is that there have been PRs that added content before those lines, shifting the content down when there were additions between last release (the time those line numbers were taken, and the current main).

> For [pvsystem.v_from_i](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.v_from_i.html) the "View on github" link directs to [these [...]

> For [pvsystem.max_power_point](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.max_power_point.html) the "View on github" link directs to [these [...]

#2434, #2433, #2435 and #2428 side effect

> For [ivtools.sdm.pvsyst_temperature_coeff](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.ivtools.sdm.pvsyst_temperature_coeff.html) the "View on github" link returns page not found.

#2256 side effect

I will have a look at it (I would expect Sphinx to link to the latest version commit SHA).

kandersolar, 2025-05-18:
> I will have a look at it (I would expect Sphinx to link to the latest version commit SHA).

It's our own code actually.  We overrode sphinx's behavior so that generated pages link to the relevant source code, even when the source code structure doesn't match the built website structure (see https://github.com/pvlib/pvlib-python/pull/913/files).

Anyway I think it should be possible to fix the issue @cwhanse encountered.  Instead of hardcoding the link to the `main` branch [here](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/conf.py#L469), we would have have some logic to point to either `main` (for `latest`) or the relevant github tag (for `stable`).  I think RTD sets an environment variable we can use to determine which of those two options is correct.

closed 2025-05-27 by kandersolar

echedey-ls, 2025-06-11:
Friendly reminder to review once a PR that modifies the number of lines gets merged.

For example, #2480 that seems to be one of the next merges:
- Stable branch is good: https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.tracking.calc_surface_orientation.html -> https://github.com/pvlib/pvlib-python/blob/main/pvlib/tracking.py#L208-L260
- Latest branch is pending for merged PR, thou link seem well-formed: https://pvlib-python.readthedocs.io/en/latest/reference/generated/pvlib.tracking.calc_surface_orientation.html -> (not linking to avoid confusion of current line numbers with latest ones once PR gets merged)