pvlib/pvlib-python #2483: Add Location Search Functionality via OpenStreetMap API

kadheer, 2025-06-16:
Enhance location lookup with city/country search instead of only coordinates.

It is additional feature which could be helpful to visit the location via input of name of the city and country.

We can use OpenStreetMap Api for this.

**1. Feature Request**
Currently, PVLib requires manual latitude/longitude inputs for location creation. This enhancement adds city/country search functionality using OpenStreetMap's Nominatim API.

**2. Proposed Solution**
```python
from pvlib import Location
loc = Location.from_query("Berlin, Germany")  # Returns Location object
```

mikofski, 2025-06-16:
IMHO converting street address to coordinates seems like something a user could do on their own independent of pvlib. For example, https://pvfree.azurewebsites.net/pvlib/ uses leaflet to do similar. Therefore, I would vote against its inclusion, as it’s not a core feature, would be a maintenance burden, and isn’t part of our mission.

AdamRJensen, 2025-06-16:
> Vice-versa can we do that. In sense it would be more user-friendly as an user can easily look for any location as soon as he enters an address. That would be more easy navigating saving one the hassle for entering coordinates all the time. @kandersolar   @mikofski 

It certainly has usage but the functionality is covered in other packages very well