pvlib/pvlib-python #2505: All-sky adjustment to clear sky

mikofski, 2025-07-24:
**Is your feature request related to a problem? Please describe.**
Recent work demonstrated spectral affected by cloud, clearly under predict shift

**Describe the solution you'd like**
Ernst develop simple connection to adjust clearly spectrum to cloudy

**Describe alternatives you've considered**
Implement Ernst correlation

**Additional context**
Use farms nit

RDaxini, 2025-07-24:
Related: #1669 (?)

kandersolar, 2025-07-25:
Check out #1201, I think it's implementing the Ernst method (although stale for 4 years now).

RDaxini, 2025-07-28:
> Check out [#1201](https://github.com/pvlib/pvlib-python/pull/1201), I think it's implementing the Ernst method (although stale for 4 years now).

I'll try to have a look over the PR some time this week. If the original contributor is no longer interested, I could take over the work. I think having some kind of cloud correction to clear sky spectral simulations in pvlib would be a valuable addition.