pvlib/pvlib-python #2506: pvlib.temperature.ross could accept ross parameter as a NOCT alternative

ramaroesilva, 2025-07-26:
**Is your feature request related to a problem? Please describe.**
Some papers on module temperature modelling propose Ross coefficient values instead of NOCT. Because this coefficient is not accepeted as input to `pvlib.temperature.ross`, users need to first manually compute the equivalent NOCT.

**Describe the solution you'd like**
Include Ross coefficient in `pvlib.temperature.ross` as an alternative input to NOCT. Table 2 in [Skoplaki and Palyvos (2009)](https://www.sciencedirect.com/science/article/abs/pii/S0960148108001353) also proposes values for different mountings that could be mentioned in the documentation (or defined as default configurations that are activated through a string input - similar to the `model` input in `pvlib.irradiance.perez`).

**Describe alternatives you've considered**
An alternative would be to only define default configurations that internally already consider the equivalent NOCT of each mounting.

**Additional context**
Table from Skoplaki and Palyvos

[image]

cwhanse, 2025-07-26:
I concur that adding the "k" parameter (Ross coefficient in Skoplaki and Palyvos) would be useful. Either `noct` or `k` would be required, and providing both should raise an error.

I also think that the reference needs adjustment. I can't locate a copy of that 1981 paper anywhere. I suspect I got the reference from some other source and didn't have the original. [Other authors](https://www.sciencedirect.com/science/article/pii/S2352484720300330) cite the model as "Ross Jr, R. G. "Flat-plate photovoltaic array design optimization." In 14th Photovoltaic Specialists Conference, pp. 1126-1132. 1980." My guess is that this same equation is presented in different formats (noct, k) in a number of places. Citing one as the basis for the function's name is sufficient. I would also add a second corroborating reference that's easier to find. The paper at the above link would suffice, IMO.

ramaroesilva, 2025-07-28:
Thanks for the input @cwhanse.

@RDaxini is going to show me the ropes on how to contribute with a PR for issue #2508. Once I'm a PR master, I don't mind taking the lead on this.

kandersolar, 2025-07-28:
> I can't locate a copy of that 1981 paper anywhere.

https://web.archive.org/web/20221122181924/https://www2.jpl.nasa.gov/adv_tech/photovol/ppr_81-85/Design%20Tech%20for%20FP%20PV%20Arrays_PVSC1981.pdf

cwhanse, 2025-07-31:
> Should the link provided by [@kandersolar](https://github.com/kandersolar) be include somehow in the function documentation? If so, how?

I don't think it's necessary to include that link. Adding a second reference to Sun et al. 2020 (or similar) would provide an easier-to-find source.

closed 2025-09-22 by kandersolar