pvlib/pvlib-python #2508: Broken link in pvlib.iotools.get_pvgis_hourly documentation

ramaroesilva, 2025-07-26:
**Describe the bug**
In `pvlib.iotools.get_pvgis_hourly`, the link for the **PVGIS Non-interactive service** (Reference [3]) is broken.

**To Reproduce**
1. Go to the function [documentation page](https://pvlib-python.readthedocs.io/en/latest/reference/generated/pvlib.iotools.get_pvgis_hourly.html) 
2. Press the hyperlink in **PVGIS Non-interactive service** (Reference [3])

**Expected behavior**
Open working webpage.

**Additional context**
Proper link [here](https://joint-research-centre.ec.europa.eu/photovoltaic-geographical-information-system-pvgis/getting-started-pvgis/api-non-interactive-service_en).

RDaxini, 2025-07-26:
Thanks for picking up on this @ramaroesilva. Would you be interested in [contributing](https://pvlib-python.readthedocs.io/en/stable/contributing/introduction_to_contributing.html) a fix in a new pull request?
We can guide you through the process if you are interested.

ramaroesilva, 2025-07-28:
Yes, please! I have some basic, more individual kind of experience with GitHub, but I've never contributed to an external repository.

RDaxini, 2025-07-28:
> have some basic, more individual kind of experience with GitHub, but I've never contributed to an external repository.

An additional comment:
The only real difference is that instead of committing/pushing to your own repositories/branches, you will fork+clone (create a copy) of the pvlib repo, create a branch for you to work on, and commit+push changes to your branch on your copy of the repo. We then compare and review changes on your branch from your copy, to the main branch from the master repo, via a pull request.

echedey-ls, 2025-07-28:
To add a bit more of context, our reference that ranges from broadest to smallest details is https://pvlib-python.readthedocs.io/en/stable/contributing/index.html
It's rare that it covers all the possible issues you encounter, but the main point is to learn the process. We'll be looking forward to that PR!