pvlib/pvlib-python #2513: Suggestion: Add pip/conda install commands to README installation section

brwerx, 2025-07-29:
README mentions that pvlib can be installed using pip and conda, it does not include the actual command-line examples. For clarity and ease of use especially for new users it might be helpful to add the following standard commands to the Installation section:

```
pip install pvlib
conda install -c conda-forge pvlib
```

This small addition could improve accessibility and bring the documentation in line with common practices for Python packages on GitHub and PyPI.

echedey-ls, 2025-07-29:
@brwerx are you interested in opening a Pull Request to fix that? We can guide you through the whole process if you want. In the docs we also have a lot of information on how to contribute, thou it's probably too extensive for what's needed here.
🔗  https://pvlib-python.readthedocs.io/en/stable/contributing/index.html

brwerx, 2025-07-30:
@echedey-ls yes I would be interested in opening a Pull request to fix this. Please let me know what I'll need to do to get things going.

brwerx, 2025-07-31:
@echedey-ls got it! I will get on this and if I have any issues I'll refer to the document. thank you!!

closed 2025-08-04 by kandersolar