pvlib/pvlib-python #2529: Add "poa_" prefix to `return_components=True` transposition model outputs

kandersolar, 2025-08-06:
> Would the powers that be be willing to make the component name’s consistent across all model functions? For example, perez returns sky_diffuse, while poa_components returns poa_sky_diffuse. It also seems that poa_circumsolar, etc. would be more consistent and descriptive for those diffuse components. (A counter argument might be than the underlying plane need not necessarily be for a solar PV array.)

 _Originally posted by @markcampanelli in [#1553](https://github.com/pvlib/pvlib-python/issues/1553#issuecomment-1931254530)_

I think the argument makes sense.  Even further: `pvlib.bifacial.infinite_sheds` returns keys with the `poa_` prefix.  

The downside is that there's not a good way to deprecate DataFrame column names (ref https://github.com/pvlib/pvlib-python/issues/1403#issuecomment-1057405773), so it would be a hard change.  It's unfortunate, but I think I'm +1 to making the change.

cwhanse, 2025-08-06:
> I'm +1 to making the change

As am I. In v0.14.

ramaroesilva, 2025-08-06:
Duly noted, thanks for clarifying. So, will split these two updates in separate branches/PRs.

adriesse, 2025-08-07:
> So: adding the parameter to `isotropic`, `klucher`, etc can be done any time, while changing the returned names in `perez`, `haydavies`, etc should wait for v0.14.0.

On the other hand, it could be seen as "cleaner" to bring them in together, otherwise there is a period where there is inconsistent naming of components.

kandersolar, 2025-08-07:
It sounds like there are three proposals so far:

1) Add prefixed names to `isotropic` et al. in v0.13.1, rename `perez` et al. to prefixed names in v0.14.0
2) Add prefixed names to `isotropic` et al. in v0.14.0, rename `perez` et al. to prefixed names in v0.14.0
3) Add un-prefixed names to `isotropic` et al. in v0.13.1, rename all functions to prefixed names in v0.14.0

Option 1 has the downside of inconsistency across functions for the period of v0.13.1–v0.14.0.
Option 2 has the downside of delaying useful functionality.
Option 3 has the downside of requiring additional breaking changes.

Here is a proposal that has none of those downsides: 

4) Add prefixed names to `isotropic` et al. in v0.13.1, add (_not rename_) prefixed names to `perez` et al in v0.13.1, and drop un-prefixed names in `perez` et al. in v0.14.0.

The downside here is that `perez` output will temporarily have duplicate columns under similar names, but that seems less objectionable than the downsides of Options 1–3.

cwhanse, 2025-08-07:
+1 for Option 4. I'm not opposed to the deprecation warning @ramaroesilva suggests but it may become onerous as these transposition functions are frequently used.

ramaroesilva, 2025-08-07:
> +1 for Option 4. I'm not opposed to the deprecation warning [@ramaroesilva](https://github.com/ramaroesilva) suggests but it may become onerous as these transposition functions are frequently used.

+1 for 4 as well. Indeed, it could become a bit cumbersome for users.

kandersolar, 2025-08-19:
> can anyone tell me the expected timing for v0.14 to come out (or better yet, where can such timings be found)?

We have been doing ~quarterly releases, but that's for releases in general, so v0.14.0 doesn't have a specific date yet.  If we decide the next release will be v0.14.0 (rather than v0.13.1) then it would be mid-September.  

Will the next release be v0.14.0?  It could be, but our last two releases were .0 releases, and I'd personally be a little reluctant for us to have three releases in a row with breaking changes.  Just my 2 cents.