pvlib/pvlib-python #2540: FS spectral correction: parameters lack a reference

cwhanse, 2025-08-21:
The first solar spectral correction function `spectral.mismatch.spectral_factor_firstsolar` provides coefficients for a single `cdte` product, see [here](https://github.com/pvlib/pvlib-python/blob/6dfeaf8ee97db6a7a8f3c8109eed8b5d285c4e41/pvlib/spectrum/mismatch.py#L253).

The coefficients are different from those found in the primary reference, the 2016 IEEE PVSC paper by M. Lee and A. Panchula [2]. 

The backstory: this python code was ported from a [function](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_FSspeccorr.m) provided to PVLib for Matlab by the primary author, Mitchell Lee. The Matlab function was updated by M. Lee with parameters from his 43rd PVSC presentation. It appears that the finalparameters did not make it into the PVSC paper.

The Matlab code has comments that include a set of parameters for FS modules older that Series 4-2, but these parameters aren't those appearing in the paper, either.

I think we need to add a new reference, private communication with M. Lee, to memorialize the source of the built-in parameters. I can't find [4] anywhere, which might contain the parameter values. [4] appears to be a non-public work product from TUV to a client.

adriesse, 2025-08-22:
@cwhanse unfortunately not. Could someone or some organization lobby First Solar to make public a document listing the coefficients?

kurt-rhee, 2025-08-22:
I can reach out to to the PlantPredict team, I know they have a contact over there.