pvlib/pvlib-python #2541: pvlib.bifacial.infinite_sheds.get_irradiance_poa does not change with height parameter

kurt-rhee, 2025-08-22:
**Describe the bug**
https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.bifacial.infinite_sheds.get_irradiance_poa.html#pvlib.bifacial.infinite_sheds.get_irradiance_poa

The output of the function above does not change when the height parameter changes.

```
                rear_irradiance_group = get_irradiance_poa(
                    surface_tilt=group_df["rear_surface_tilt"],
                    surface_azimuth=group_df["rear_surface_azimuth"],
                    solar_zenith=group_df["apparent_zenith"],
                    solar_azimuth=group_df["azimuth"],
                    gcr=gcr,
                    height=10000000,
                    pitch=pitch,
                    ghi=group_df["ghi"],
                    dhi=group_df["dhi"],
                    dni=group_df["dni"],
                    albedo=ALBEDO,
                    model="hay_davies",
                    dni_extra=group_df["dni_extra"],
                    iam=1.0,
                    npoints=6,
                    vectorize=True,  # Vectorize is fine for time-series inputs
                )
```

**To Reproduce**
Steps to reproduce the behavior:

Some sample data
```
string_id met_name                      time  surface_tilt  surface_azimuth         ghi        dhi         dni  apparent_zenith    azimuth    dni_extra  pitch  racking_equipment_id  racking_controls_gcr  module_equipment_id  bifaciality_factor  pile_height  rear_surface_tilt  rear_surface_azimuth
122780        429       05 2025-08-18 07:15:00-06:00     11.563849        90.000000   64.606815  23.909445  482.167134        85.023666  77.323903  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         168.436151            270.000000
122781        429       05 2025-08-18 07:20:00-06:00     13.798609        90.000000   79.229623  26.120372  526.216676        84.090495  78.089598  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         166.201391            270.000000
122782        429       05 2025-08-18 07:25:00-06:00     16.098271        90.000002   94.731742  28.557150  563.347112        83.150932  78.853213  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         163.901729            270.000002
122783        429       05 2025-08-18 07:30:00-06:00     18.475625        90.000000  117.185185  39.581374  579.055416        82.206119  79.615158  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         161.524375            270.000000
122784        429       05 2025-08-18 07:35:00-06:00     20.947126        90.000002  136.620012  46.861257  596.118306        81.256884  80.375843  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         159.052874            270.000002
122785        429       05 2025-08-18 07:40:00-06:00     23.533385        90.000000  158.033562  55.620140  612.808995        80.304018  81.135684  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         156.466615            270.000000
122786        429       05 2025-08-18 07:45:00-06:00     26.261737        90.000000  172.369276  50.339806  664.439619        79.348009  81.895102  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         153.738263            270.000000
122787        429       05 2025-08-18 07:50:00-06:00     29.168602        90.000000  190.162325  53.228972  684.073245        78.389286  82.654525  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         150.831398            270.000000
122788        429       05 2025-08-18 07:55:00-06:00     32.304751        90.000001  215.508648  69.363785  674.533014        77.428165  83.414384  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         147.695249            270.000001
122789        429       05 2025-08-18 08:00:00-06:00     35.744402        90.000001  226.806553  59.080368  719.509943        76.464956  84.175121  1327.346081   5.05                     8                   0.4                   12               0.791          1.6         144.255598            270.00000
```

**Expected behavior**
A clear and concise description of what you expected to happen.

I would expect that a higher height would give a larger irradiance value

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``:  0.13.0
 - ``pandas.__version__``:  2.2.3
 - python: 3.12

**Additional context**

cwhanse, 2025-08-25:
That function returns POA irradiance on one face, not the sum of front and back. The face is implied by `surface_tilt`. So I suspect you are seeing only front side irradiance.

What do you get if you use `infinite_sheds.get_irradiance`?

Suggestions welcome for improving those docstrings.

cwhanse, 2025-08-25:
Hmm. I can't see a disconnect in the tilt and azimuth part, but there's this:

```
                    pitch=height,
```
Intended?